tanakamasayuki/EspUsbHost #9: Mouse "Disconnect" (or else is ignored by the host)?

coofercat, 2024-05-31:
I'm using the [example code](https://github.com/tanakamasayuki/EspUsbHost/blob/master/examples/EspUsbHostMouse/EspUsbHostMouse.ino) to show a mouse moving. I've removed the serial printing so it doesn't slow anything down, and instead flash the onboard LED if movement is detected.

For reasons I can't explain, every once in a while, the mouse seems to "disconnect" or otherwise becomes ignored by the host. My mouse has a light which flashes when it is genuinely disconnected, or is connecting to a host, and this doesn't flash, suggesting this isn't a proper USB-level disconnect. The repeated calls to `usbHost.task();` in the `loop()` are definitely being made.

The only solutions I've found are to disconnect and reconnect the mouse, or else reset the ESP.

I've tried increasing the logging for the underlying ESP libraries (in the Arduino UI: Tools -> Core Debgging Level -> Info), but don't get anything from that when this problem occurs. This logging DOES show the mouse being disconnected/reconnected, again suggesting this isn't actually a USB-level disconnect.

Other that that, I don't have a lot of information to go with. It's hard to reproduce, because it can be hours before it happens, and it doesn't seem to matter what you're actually doing with the mouse at the time either. I'm happy to experiment a bit if anyone has any ideas what might be worth trying.

alex-k8, 2024-10-26:
Any progress on this? When it happens I notice that the usb_host_client_handle_events just times out and returns nothing, basically the mouse stops sending data.

alex-k8, 2025-10-24:
> Im still noticing this issue, did anyone find a solution to this?

Unfortunately not. If we're talking about the same issue, I had opened an issue in the official esp-idf repo, and I got some response back from the devs that they managed to reproduce it, and had some workarounds, but I couldn't get it to work personally. It might be worth looking at that and commenting as well.
https://github.com/espressif/esp-idf/issues/14996