tanakamasayuki/EspUsbHost #12: Using ESP32-S3-DevKitC-1 cannot connect to USB keyboard

creepscompilation, 2024-11-21:
I have two different development boards that I am using with this library and the basic "EspUsbHostKeybord" sample code.
ESP32-S3-DevKitC-1 and ESP32-8048S070N/C 7-Inch touchscreen with the ESP32-S3-WROOM-1.

Both library and code install and compile without issues.
I also confirm the wiring is accurate, however, I confirm that the USB keyboard does not enumerate.

I am compiling with using the "ESP32S3 Dev Module" in Arduino.

Question: Do I need to hard wire any other pins?

creepscompilation, 2024-11-21:
I didn't see a wiring diagram so, here is my confirmed wiring with continuity check.
USB- Female Socket 
Pin # 1 = +5vdc
Pin # 2 = D-  = ESP32-S3-WROOM-1 (GPIO19) =  PIN # 13 on the ESP32-S3-WROOM-1
Pin # 3 = D+ = ESP32-S3-WROOM-1 (GPIO20) =  PIN # 14 on the ESP32-S3-WROOM-1
Pin # 4 = Ground

I am not sure if I need to pull-up or pull-down another PIN for HOST to work, I can't find
any documentation for that. I am assuming the library and example code should work 
with the wiring I am using, but I must be wrong.

creepscompilation, 2024-11-21:
I got it working by shorting the USB-OTG pins on the bottom of the module, however, does that mean I
must use a USB-C connector with an adaptor to plug in USB-A devices? It appears to be a hardware issue.

How do I get a standard USB-A (4-pin) USB keyboard to work without the need for a USB-C to USB-A adapter?

I found the schematic for the ESP32-S3-N16R8-N8R2 board that I am using.
https://lededitpro.com/esp32-s3-development-board-2-4g-wifi-module/

creepscompilation, 2024-11-21:
Is there a schematic circuit for this project that shows how to wire a standard USB-A (4-pin) USB keyboard?

hpwit, 2025-01-15:
hello I have plugged my keyboard using an usb adapter  but I only get back e m a when typing and only a certain keys.
I am using a corsair keyboard
and a 7inch module from maker labs. I have seen this working on a video

nak435, 2025-06-27:
I also connecting it to D-:GPIO19 and D+:GPIO20, but it didn't work at first. 
So I measured the VCC-GND voltage of the USB-A female, and it was only 4V. 
I prepared another 5V and applied it to the USB-A female, and the keyboard worked without any problems.
OTG is not required, but GND must be shared.

@creepscompilation 

How about measuring the voltage?

9dl, 2025-07-10:
My Pinout
```
| Component    | Signal    | ESP32-S3 Pin     |
| ------------ | --------- | ---------------- |
| USB Breakout | D+        | GPIO20           |
| USB Breakout | D−        | GPIO19           |
| USB Breakout | VBUS      | 5V from PD board |
| USB Breakout | GND       | GND from PD board|
```

My Code
```cpp
#include "EspUsbHost.h"

class MyEspUsbHost : public EspUsbHost {
    void onKeyboardKey(uint8_t ascii, uint8_t keycode, uint8_t modifier) {
        if (' ' <= ascii && ascii <= '~') {
            Serial.printf("%c", ascii);
        } else if (ascii == '\r') {
            Serial.println();
        }
    };
};

MyEspUsbHost usbHost;

void setup() {
    Serial.begin(115200);
    delay(500);

    usbHost.begin();
    usbHost.setHIDLocal(HID_LOCAL_Japan_Katakana);
}

void loop() {
    usbHost.task();
}
```

Serial:
```
E (22873) HUB: Root port reset failed
E (23133) HUB: Root port reset failed
E (23393) HUB: Root port reset failed
E (23653) HUB: Root port reset failed
```

Anyone knows what i could do to fix this issue?